sfuphantom/firmware #11: Phantom CAN Driver

gabesoares, 2019-12-09:
Create Team Phantom driver for CAN

phantom_can.h:
- Node defines (BMS, VCU, Dashboard, Charger, Inverter), 
- CAN message IDs, defines, API function prototypes

phantom_can.c
- Implement API functions (CAN_send(message), can receive message, can_initialize,

Junaidk11, 2019-12-12:
CAN Node 1(VCU)  and  Node 2(BMS)  are functional. Troubleshooting addition of Dashboard (Node3) to the network.

Junaidk11, 2020-01-19:
Node 3 ( Dashboard) is functional when there is either Node 1 or Node 2 in the network,  i.e. with all three nodes in the Network, the CAN functionality is not reliable, as sometimes some of the nodes are unable to receive the messages.

Junaidk11, 2020-02-07:
Node 3 (Tiva) is able to transmit to other nodes on the network, Unable to receive data from either of the Nodes - I believed ISR for receive interrupts need to be looked into.

Junaidk11, 2020-02-09:
Further inspection led to the conclusion that all three nodes transmit messages on the bus. However, only two out of the three nodes filter the messages on the CAN bus.